hamish2014/FreeCAD_drawing_dimensioning #12: Windows Install Instructions

BPLRFE, 2015-01-08:
Tested with 015.4415 Development Snapshot on a Windows 7 64bit-System

- Download repository as ZIP
- Assuming FreeCAD is installed in "C:\PortableApps\FreeCAD 0_15"
- go to "C:\PortableApps\FreeCAD 0_15\Mod" within Windows Explorer
- create new directory named "DrawingDimensions"
- unzip downloaded repository in "C:\PortableApps\FreeCAD 0_15\Mod\DrawingDimensions"
- when starting FreeCAD you will now have a new workbench-entry called "DrawingDimensions"

hamish2014, 2015-01-12:
Thanks for the instructions BRLRFE;
Before I add them, could you clarify how you went about installing PySide and Numpy?

BPLRFE, 2015-01-12:
Pyside and Numpy are integrated in the dev-Snapshots 0.15 (Thanks to User SGROGAN).
AFAIK he has ATM Problems with also integrating scimpy, but that's another Story.
I did as described above, nothing else to worry about.
Maybe you should add a general note that your app only runs with FreeCAD 0.15 and (hopefully)
above

hamish2014, 2015-01-12:
Having pyside and numpy included is very convenient indeed, that is excellent.
Anyway your instructions have been added, thanks for the info!

closed 2015-01-12 by hamish2014