hamish2014/FreeCAD_drawing_dimensioning #13: Display Grid and make dimensions/Text snap to grid

BPLRFE, 2015-01-08:
Would be nice to toggle on/off display of grid and have dimensions and Text snap to grid points.
Maybe you can use the same grid as in the draft workbench ?

hamish2014, 2015-01-12:
Hi, I do not really know much about the draft workbench, but I suspected this would be a difficult task.
I say this because, the primary FreeCAD graphics view is rather different from the QT graphics view which the drawing and drawing dimensioning workbenchs use.
For this reason, i suspect it may be easier to program a grid from scratch scratch then port it from the draft workbench.

Either way, i am labelling this issue an enhancement.
If i do end having time to attend to it, it will be in the distant future.

hamish2014, 2015-09-04:
Grid feature added in latest commit