hamish2014/FreeCAD_drawing_dimensioning #19: Font size in px

triplus, 2015-01-14:
Hi.

Added text (add text or dimension text) currently does not add unit to font-size style attribute. Adding unit px would correct font rendering of exported drawing in for example Firefox.

Cheers!

hamish2014, 2015-02-17:
support added

hamish2014, 2015-02-17:
specify font size not working, there is no difference between
* font-size="3.6pt"
* font-size="3.6cm"
* font-size="3.6"px
any ideas?

triplus, 2015-02-17:
Hi.

I tested new font size and font family preference and it works good.

Great work!

About the mentioned issue i would say this is QtSvg rendering engine related limitation. If you export the drawing you will see other SVG rendering software like Inkscape/Firefox will use the provided unit. Currently in FreeCAD Drawing Workbench there is a rule 1px = 1mm therefore usually everything (including drawing template text) is defined in px. From this feature request point of view i was thinking FreeCAD Drawing Dimensioning Workbench should always add px as unit to text size. The reason was Firefox didn't render text correctly if px unit wasn't set BUT in latest Firefox 35 it looks like this was fixed!

Therefore i suggest you remove any mention of font UNIT from preferences and Add text dialogue and close this feature request as implemented. If for whatever reason user would need to explicitly add px as a unit in addition to size it can be done now. If in the future different SVG renderer (like WebKit) will be used in FreeCAD then support for units in your workbench is implemented and can be used!

hamish2014, 2015-02-18:
Glad that font-size units work in other SVG rendering software.
Will do as recommended; functionality left the same, except all mention of units is removed.
i.e. default font-size changed from `3.6pt' to `3.6' (This will also update the addText dialog)

closed 2015-02-18 by hamish2014