hamish2014/FreeCAD_drawing_dimensioning #25: Bug with Escape Dimensioning

quick61, 2015-01-15:
There looks to be a bug when using the Escape Dimensioning function. When in the middle of creating a dimension, you decide to cancel it by using the Escape Dimensioning function the workbench stops functioning and any further attempt to add a dimension results in the following error - 

Traceback (most recent call last):
  File "/home/mark/.FreeCAD/Mod/Dimensioning/selectionOverlay.py", line 21, in mousePressEvent
    self._onClickFun( event, self, self.elementXML, self.elementParms, self.elementViewObject )
  File "/home/mark/.FreeCAD/Mod/Dimensioning/linearDimension.py", line 22, in selectDimensioningPoint
    previewDimension.initializePreview( dimensioning.drawingVars, clickFunPreview, hoverFunPreview )
  File "/home/mark/.FreeCAD/Mod/Dimensioning/previewDimension.py", line 65, in initializePreview
    preview.SVG.update()
RuntimeError: Internal C++ object (PySide.QtSvg.QGraphicsSvgItem) already deleted.

After the first failed attempt, all the vertex markers are shifted off center and no dimentions can be added.

The only way to recover is to shutdown FreeCAD and restart the session. 

FreeCAD version - 

OS: Ubuntu 14.04.1 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.15.4484 (Git)
Branch: master
Hash: f72c802f5daa378298ffbfc3f3a17ec2d35c1816
Python version: 2.7.6
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17-dev

sgrogan, 2015-01-15:
Replicated on Windows. Same behavior.

OS: Windows 7
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.15.4435 (Git)
Branch: master
Hash: f24e216db4fb923b2571ad9d856f934e81bec797
Python version: 2.7.8
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 6.7.1

hamish2014, 2015-01-18:
Hi Folks,
Escape dimensioning should be fixed now.
Aftering pulling the last version, is the issue fixed on yourside?

quick61, 2015-01-18:
YES! Yes it does all seem to be working here now. The Esc function is functioning as expected. 

Thanks!

OS: Kubuntu 14.04.1 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.15.4498 (Git)
Branch: master
Hash: 86882eec6a7f862873c2ef1948bf0c539f2e9e5b
Python version: 2.7.6
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17-dev

hamish2014, 2015-01-20:
Pleasure quick61, closing bug

closed 2015-01-20 by hamish2014