hamish2014/FreeCAD_drawing_dimensioning #29: feature request - groove symbols

microelly2, 2015-01-21:
is it possible to extend the notation indicator  to groove symbols?
http://www.cad-programme.com/tipps/schweisssymbole.html

hamish2014, 2015-02-26:
Hi microelly2,

I have added all the grove symbols at the link mentioned.
Could you please pull the latest version and tell me what you think.
The grove symbols are scaled according to the font size.

[image]

hamish2014, 2015-02-26:
Correction, i did not add the last symbol on the list.

sgrogan, 2015-03-06:
Thanks hamish2014 for this!
It works perfectly on kubuntu!
On windows I get "Unhandled unknown exception caught in GUIApplication::notify."
The workbench loads, but without the groove icons in the toolbar.
I tried coping the resources folder to different places, like my macro directory (...\Appdata\roamig\FreeCAD\
With no luck.
Everything was a treat until the last git pull.

hamish2014, 2015-03-11:
Hi sgrogan,
Just wondering if you have come managed to make progress with your windows issue?

sgrogan, 2015-03-11:
Hi hamish,To be honest I haven't been able to look seriously. I hope to have time this weekend and I will report anything I find. Thanks for the interest.And again thanks for your efforts. You really have added a lot of functionality with this. If you do a google search on the FreeCAD help forum you will see a lot of frustrations with FreeCAD relieved because of you.Chris-----Original Message-----
From: hamish2014 
Sent: Mar 11, 2015 5:00 AM
To: hamish2014/FreeCAD_drawing_dimensioning 
Cc: sgrogan 
Subject: Re: [FreeCAD_drawing_dimensioning] feature request - groove symbols (#29)

Hi sgrogan,
Just wondering if you have come managed to make progress with your windows issue?

microelly2, 2015-03-13:
Hi hamish,  looks nice,
there seem to be some problems with windows ?!
http://forum.freecadweb.org/viewtopic.php?f=13&t=10044&start=20#p81325

sgrogan, 2015-03-15:
@hammish
I've made some progress
x64 windows I get the previous error and no welding icons. FreeCAD crashes if I switch to a python wb or any C workbench that has not been previously activated in the same session.
If I remove the comment from line 68 of InitGui.py the crash is fixed and I get the X blank icons for the welding symbols and a warning like
"Cannot find icon: C:\Users\Chris\Desktop\FreeCAD_Portable\Mod\Drawing-Dimensioning\Resources\icons\welding\square_groove.svg"
for all the welding icons, although this is the correct full path. The tooltips work for the icons though.

In x32 windows I get no errors at workbench load, but the same cannot find icon warning. The behavior is the same after cuncommenting the InitGui line. (I do get the drawline icon in both cases and it seems to work)

I haven't checked what  uncommenting this line does on unix.

hamish2014, 2015-03-17:
@sgrogan , 

Thanks for your help!

The issue appears to do with welding svg icons.
These icons were generated in weldingSymbols.py, without sufficient svg parameters for Windows system.
After adding extra svg headers, everything is now working on my windows 7 x86 boot, using the FreeCAD devoplment built posted on source forge on 15-03-2015.

Please pull the latest version of the drawing dimensioning workbench and let me if know if it works.

Regards,

sgrogan, 2015-03-17:
I tested x86 and x64 builds on Win7 x64 computer
and x86 build on XP x64 computer (after swapping out numpy see [url]http://forum.freecadweb.org/viewtopic.php?f=10&t=8395&start=60#p77413[url]

All work perfectly! 
I think you can close this. Thanks again for the effort!

hamish2014, 2015-03-18:
Pleasure @sgrogan , thanks again for help fixing the bug.

closed 2015-03-18 by hamish2014