hamish2014/FreeCAD_drawing_dimensioning #30: Distance Between Parallels

quick61, 2015-01-28:
Ran into this today. Needed to dimension the distance between 2 parallel lines of different length that were on an angle. Could not get the proper dimension. Can provide a link to file if needed.

quick61, 2015-01-28:
[image]

hamish2014, 2015-01-29:
Yes it would be possible, you will have to wait a while though;
The list of feature requests for the drawing dimensioning wb has gotten long, and it would not be fair if i jump the queue.

quick61, 2015-01-29:
No problem. Very much appreciate the effort. Just ran into the limitation working on a question in the FreeCAD help forum and figured it should be mentioned.

quick61, 2015-02-13:
I was just thinking about this, and maybe just adding a midpoint to lines would work for most all instances. Just a thought for after you get the other request added in. All of them look like they would be most welcome additions.

hamish2014, 2015-02-18:
Hi quick61, please pull the latest version and check the distance between parallel functionality.
The functionally has been added to the linear dimension command;
select a line, and lines parallel to that will also be selectable ...

[image]

quick61, 2015-02-18:
Oh yes! Very nice. This is working great. Thank you. I would say this feature request could be considered closed.

hamish2014, 2015-02-18:
Pleasure, closing issue.

closed 2015-02-18 by hamish2014