hamish2014/FreeCAD_drawing_dimensioning #37: tuple assignment error

lanciaux, 2015-04-14:
I don't yet know exactly how - I'm not writing on Linux machine it happened but since you have added folding icons - it brings after a not so small delay "tuple assignement error" when creating the folding bar. I do not know if python 3.x is installed on that particular machine. On the one I'm writing, Python 3.4 is installed and everything running fine.

Bye the way, your additions are really great - I will put a post in Assembly2 concerning a regular small issue I have (maybe bad understanding of me)

hamish2014, 2015-04-15:
Hi  lanciaux,

This seems like a bug. 
Does the tuple assignment error give more information, such as a line no. ? 
(the Linux system i am on uses Python 2.7)

hamish2014, 2015-04-15:
I have have only used the unfold command on simple geometries;
Perhaps the error originates from the shape you are attempting to unfold.

Therefore, please send me the FreeCAD file which is causing the problem;
Put it on google drive or drop box, and send me the link.

lanciaux, 2015-04-15:
Hello, unfortunately no line error or I would have given it. Here are some more information I have hoping this will help :

1. Python 2.7.8 is also installed at least one on one of Linux it happens ... so maybe not an issue of Python version ... at least directly ;

2. On both machine it happens, these are BUILDs i do on a regular basis [2 times a week more or less] (from eitheir sf.net, either github.com) - on the Linux it runs perfect, this is a build collected from PackMan depot (I'm on Suse 13.1 or 13.2 - the one working is on 13.2, I have failed in 13.1 and one failing on 13.2) ... so this is not OS issue (coincidence) ;

3. I have both Assembly2 and Drawing_Dimensions installed from you (as possible as often the latest version) ;

4. It happens with an empty model as well because it fails ONLY when creating folding bar icon with 2 or icons (unfold, unfold info + dxf export) ;
 
5. A more interesting fact is maybe this : on one of the failing linux, opening menu toolbar drawing dimensions fails ... no icons appear at all, then I open assembly2 toolbar which open fully, then going back to drawing dimensions toolbar, I have the first part of drawing dimensions icons but still no folding icons neither welding icons : is it due to a name of icon folding bar already assigned in Assembly2 ? What about some kind of namespace trouble variable in a global package ?

6. I'm new to Python and have only a user knowledge of FreeCAD as now 

Yours

Ludovic (Belgium)

hamish2014, 2015-04-16:
Hi Ludovic,

Thanks for all the info.
Unfortunately, I am still not able to be of any help.
Perhaps, someone on the FreeCAD forum can help.
Please let me know if you figure out what the problem is.

Good luck!

lanciaux, 2015-04-17:
[image]

lanciaux, 2015-04-17:
At least something to dig up :
1. quick typo in file In file InitGui.py : line 89 ... 'drawingDimensioning_exportToDxf', ... no "," at the end of the line because liste close after ;
2. I have debug "old fashion way" : in file unfold.py - it fails here :

"try:"
"    from matplotlib import pyplot"           =>>>>> this line fails with "tuple assignment ..." but I have matplotlib installed !
"except ImportError:"
"    pyplot = None"                                  =>>>>> If I force : it works !

hamish2014, 2015-04-18:
Bravo lanciaux,
I used matplotlib for debugging purposes only.
Given the library does cause problems on certian systems, it has now been remove it.

closed 2015-06-11 by hamish2014