hamish2014/FreeCAD_drawing_dimensioning #38: DXF export

triplus, 2015-04-15:
Hi.

Nice work on Unfold faces tool. 

Anyway i have one question about DXF export. How come you didn't just use built in support (File -> Export -> Autodesk DXF)? Current solutions needs Inkscape installed and FreeCAD run from command line and only works on Linux.

Wouldn't it make more sense to just use FreeCAD and its default exporter?

hamish2014, 2015-04-15:
Yes, using internal FreeCAD functions would be preferable.
Are you able to successfully export a drawing page using File -> Export -> Autodesk DXF?
I am having some difficulty on my side, what steps exactly did you follow?

triplus, 2015-04-15:
Yes it does work but appropriate libraries have to be downloaded first:

https://github.com/yorikvanhavre/Draft-dxf-importer

Current FreeCAD policy is to include only LGPL software and no GPL by default. Mentioned DXF import-export libraries were downloaded by default on DXF import-export operation in the past but Debian didn't like that and FreeCAD changed in a way user has to manually enable (Opt-In) the download. 

Open Draft workbench and after that select Edit -> Preferences. Under Import-Export -> DXF / DWG tab enable Automatic update. After that FreeCAD will download mentioned libraries on first DXF import-export operation and it should work. If it does not work restart FreeCAD and try again.

RoussetA, 2015-09-25:
Hi,
The DXF export works well but the dimensions created with the drawing dimensioning workbench does not appear on the exported dxf file when i try open it with Rhino. When opening it with Freecad nothing happens. Did i miss something or is it a known limitation?

hamish2014, 2015-09-25:
Have you tried qcad? Below is a screen shot of a dxf i exported this afternoon using the Drawing Dimension workbench:
[image]

I won't lie the dxf exported is very limited, but it is suffient for my purpose:
i.e. quickly creating dxf files to send to a laser cutting and bending company ...

RoussetA, 2015-09-28:
Dear Hamish, thanks for your feedback.
I wanted to be sure that it was not coming only from me! ;-)
No problem i'll deal with these limitations.

lrl1956, 2015-12-11:
Hamish, I too am having trouble using the "DrawingDimensioning" dxf export, and I think it would be very beneficial to me, especially if dimensions go with it.  Below is the Errors that popped up when I tried to export...any help would be appreciated! I'm by no means a programmer, but I am really getting pumped about the usefulness that FreeCAD is becoming!
Thank you again!

dxf_export: Warning export of ?xml elements not supported, ignoring...
dxf_export: Warning export of svg elements not supported, ignoring...
dxf_export: Warning export of !-- elements not supported, ignoring...
dxf_export: Warning export of g> elements not supported, ignoring...
dxf_export: Warning export of rect elements not supported, ignoring...
dxf_export: Warning export of g elements not supported, ignoring...
Running the Python command 'dd_exportToDxf' failed:
Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.15\Mod\DrawingDimensioning\unfold_export_to_dxf.py", line 20, in Activated
    export_via_dxfwrite( dxf_fn, V)
  File "C:\Program Files\FreeCAD 0.15\Mod\DrawingDimensioning\unfold_export_to_dxf.py", line 104, in export_via_dxfwrite
    drawing.add(dxf.text( t.text, insert=(x, yT(y)), height=t.height()*0.8, rotation=t.rotation, layer='TEXTLAYER', color=color_code) )
  File "C:\Program Files\FreeCAD 0.15\Mod\DrawingDimensioning\svgLib_dd.py", line 94, in height
    return self.textRect().height() *0.6
  File "C:\Program Files\FreeCAD 0.15\Mod\DrawingDimensioning\svgLib_dd.py", line 84, in textRect
    sizeInPoints = self.convertUnits(self.font_size.strip())
  File "C:\Program Files\FreeCAD 0.15\Mod\DrawingDimensioning\svgLib_dd.py", line 81, in convertUnits
    return float(v)*factor

could not convert string to float:

hamish2014, 2015-12-14:
Hi lrl1956,

One the svg text elements is causing the dxf export function to crash.
My guess is that this text element is probably from the page template.
Would you mind uploading the FreeCAD file in question (on filedropper of the like) so that i can have a look at it.

In the meantime, I have added exception handling code so that errors of this nature won't bomb out the whole process.

lrl1956, 2015-12-17:
[Arm.txt](https://github.com/hamish2014/FreeCAD_drawing_dimensioning/files/65652/Arm.txt)
Hi Hamish2014, Thank you for your help!  I'm not real sure about the filedropper type deal, but the Arm.txt file is a file like that has given an error. I hope replacing the extension will allow you to look at the file.  Again, Thank you!  I'll be away from all forums until Monday, 12/21/2015