hamish2014/FreeCAD_drawing_dimensioning #39: Dependency on FreeCAD_assembly2

normandc, 2015-04-26:
Since adding lib3D.py and lineSearches.py symlinks, the Drawing Dimensioning workbench will not load without FreeCAD_assembly2 present. Switching to Drawing dimensioning using the workbench switcher brings the error dialog "No module named lib3D".

Would it be possible to allow the module to load by disabling the functionality provided by the assembly2 module?

The other problem is if the symlink has a hardcoded path to ../FreeCAD_assembly2/lib3D.py. If the end user has renamed the Assembly2 folder, the symlink is broken (yet the drawing dimensioning wb loads normally).

There may also be a problem with Windows users, as this path is typically used on Linux, not on Windows.

hamish2014, 2015-04-27:
Ok, let me see if i can remove the symlink

plaes, 2015-05-01:
There's still another issue: `from lineSearches import quadraticLineSearch` where lineSearches is imported from Assembly2

hamish2014, 2015-05-04:
Thanks, will fix now.

closed 2015-05-09 by hamish2014