hamish2014/FreeCAD_drawing_dimensioning #43: circle not recognized

herrdeh, 2015-05-10:
At the file attached, in one view circles seem not to be recognized by the tool. When clicking the "diameter", "radius" or "midpont" button, the small, inner circles are not highlighted.

[image]

Here's the file:
http://verkehrsplanung.com/FreeCAD/2-Kugel.fcstd

Love your tool !!!

BlixtMaskin, 2015-06-11:
Same problem, done a shape with a radius of 25 mm and it won´t be recognized, just a small circle appears when trying the radius or the circle tool.

[image]

hamish2014, 2015-06-11:
Hi, please upload the drawing pages SVG (the drawing workbench has an export option).
The SVG will help to test the selection overlay code.

BlixtMaskin, 2015-06-11:
Ok, here it goes.

http://www.filedropper.com/ovalshape

hamish2014, 2015-06-11:
Got it, thanks.
When i recommend you upload the SVG, i thought you could just upload the SVG here on github.
Unfortunately, i see now that SVGs are not supported.

Anyway, i will see if can look at the issue sometime this weekend or next.

hamish2014, 2015-06-13:
Hi @BlixtMaskin,
Try out the latest version of the drawing dimensioning workbench, and let me know if your issue is resolved.

BlixtMaskin, 2015-06-15:
Looks far better, great job! Just some minor design problems when you place the arrow around the circular shape (the arrow disappears) left, but it works.

Thanks!

herrdeh, 2015-06-16:
Still seems not to be working with my example linked above. On the left drawing, the inner (small) circles are not recognized.

hamish2014, 2015-06-16:
Hi @herrdeh, please upload the drawing in question's SVG (the drawing workbench has an export option), and I will see what I can do.

herrdeh, 2015-06-17:
'Sorry, it took me some time to understand, that your tool is working on the svgs.
Her we go:
http://verkehrsplanung.com/FreeCAD/Kugel.svg

Cheers!

hamish2014, 2015-06-18:
Thanks Herrdeh,

I have looked at the Svg and believe I have identified the problem.
The circles are not being identified because they are in fact slightly elliptical.
In particular, the Svg code ( `  <path d="M -0.751006,0 A 3.01146,3 -180 0 0 -3.76247,-3" />` ) indicates that the x-radius is **3.01146** and the y-radius is **3**.
This slight difference between the x-radius and the y-radius leads to the drawing dimension workbench identify the path as elliptical, and circular.

herrdeh, 2015-06-18:
Hi, the holes to be drilled into the body are nowhere rectangular to the surface, but ever in an angle. So in fact, at the surface the are ellopsoids, but the drills are circular.

I think, this case can appear quite often. Look at these examples:
http://verkehrsplanung.com/FreeCAD/Beispielwuerfel.svg
http://verkehrsplanung.com/FreeCAD/Beispielwuerfel.fcstd

Very straightforward part - and I guess, most people would like to apply diameters to the drillings.

Cheers,
Wolf