hamish2014/FreeCAD_drawing_dimensioning #45: Enhancement: "Invisible lines" attributes should be editable

herrdeh, 2015-05-10:
The gray, dotted lines, showing non-visible parts of my drawing, are hardly visible at my system. I'd like to be able to set their thickness, colour and line-gap-ratio, as I can do with center lines in the prefs.

hamish2014, 2015-05-21:
Hi, 

I have had a similar desire myself.
Unfortunately though, this issue has to do with drawing workbench, and not the drawing dimensioning workbench.

closed 2015-05-21 by hamish2014

arkticrainbow, 2015-08-30:
And is there any topic currently dealing with this issue (maybe on FreeCAD forum)? I also find it frustrating not to be able to change design of hidden lines. After printing its not really clear whats hidden and whats visible line anymore.