hamish2014/FreeCAD_drawing_dimensioning #46: number of digits doesnt follow prefs

herrdeh, 2015-05-10:
At prefs -> general -> units I set "1 digit after decimal point".
The tool ever provides three digits on my machine.
Bug? - Or should there be a separate control in the tool's prefs for that?

Sorry for flooding you with issues.

hamish2014, 2015-05-21:
Hi again Herrdeh,

I have marked this issue as an enhancement, maybe one day i will get round to it.

Finally thanks for posting all the issues.
You need to keep in mind, that i created the workbench to assist me with the FreeCAD work i do.
Unfortunately my job does not involve maintaining the drawing dimensioning workbench.
Therefore I do not have the time to give the workbench the polish it deserves ...

herrdeh, 2015-05-23:
Thank you very much for your work anyway. Only it made FreeCAD really useable for me, because I couldnt deal with the other ways of dimensioning.

closed 2015-06-11 by hamish2014