hamish2014/FreeCAD_drawing_dimensioning #49: Dimensioning from an origin / Bearing Table

fw0stahl, 2015-06-20:
Hello,

i use FC for my little CNC production. Mostly i have parts for milling with many bearings or turning parts with many paragraphs? (i don't know the word for it in English). So it would be really useful if i can dimensioning the parts from one origin. I've got a picture where you can see what i mean.

http://www.scritub.com/files/limba/germ ... age103.jpg

A bearing-table like this could be useful too.

http://help.solidworks.com/2013/german/ ... 4901.image

Maybe this could be implemented? :)

Kind Regards, thank you for FC :)

mk2

Post: http://forum.freecadweb.org/viewtopic.php?f=8&t=11421

hamish2014, 2015-06-21:
Hi mk2,

Thank you for your message.
I am unable at this moment to assist with your requested enhancement.
Perhaps in the future, i will get round to attending to it.

Kind Regards,
Hamish

arkticrainbow, 2015-08-30:
I would like to also "vote" for this feature to be implemented. In our shop floor we are having large tables (600x1200mm) with sometimes 30+ (even 50+) bored and drilled holes which isnt easy to dimensioning without using such tables.