hamish2014/FreeCAD_drawing_dimensioning #50: Linear Dimension, measured value width to big

fw0stahl, 2015-06-24:
Hello,

still i try to measure a new CNC part. 

1) So i want to decide if i want to sketch a line besides the arrows, because my measured value width is to big for the distance between the arrows.

http://solidworks.cad.de/images/faq/faq-bem-002a.gif  <- Example dimension 12,5.

2) I want to have "," instead of "." i'm from germany, so we use the "." for inch-measured-values.

3) I want to decide if the arrows look from inside to outside OR from outside to inside. This is important for small parts with many measured values in it.

http://www.rs1kronach.de/cad/tutorials/E3/data/funktion.jpg

4) If there is a chain dimension, i want to make a point instead of an arrow on the contact point. The start and endpoint have to show an arrow! Maybe you can create a new linear chain dimension. You can use the linear dimension tool, you only have to change that the last green point which you have choosen, has to be dialed-in for the next dimension.  So for the second, third, forth... dimension you only have to take the second green dot.

5) Automatic center line of holes. You can look how gib is the diameter and can decide how big must be the center line of the hole. Would be spare much time :)

6) Face measuring objects like the welding ones, would be useful.

Thank you for your tool, without it, it wouldn't be practical to use FreeCAD for production! So thank you very much! These comments are only incitations ;)

hamish2014, 2015-06-25:
Thanks for the suggestions fw0stahl.
I will see what i can do.
Glad you are finding the workbench useful:)

fw0stahl, 2015-07-13:
Hello,

your last changes are useful :-) A person asked in the forum for moving dimension-lines, maybe you can write the code for it? It would be useful too, if i can set the "comma"-on option in the global preferences. Instead of i have to set it every time again. For point 8) with the 10 and 7mm from the part away, if it is possible to write it, it would be awesom if you can set an option for it. For example option a) continius pulling the dimension line b) stepped dimension line setting (10, 17, 24, 31mm ... far away from the object.)

Thank you for your willingness

PS: Are you from german? I saw some welding-svg-files with german labels...

fw0stahl, 2015-07-14:
I can't set the comma option for the tolerance.

hamish2014, 2015-07-14:
Hi fw0stahl,

I will look into fixing the comma issue on the add tolerances.
Regarding setting `use comma' as default, use the bottom of the task panel `set as default'. i.e.
> click linear dimension
> check comma
> press `set as default' buttom 

PS: i am not German (lol :), the person who requested the welding features is however (i suspect). I am South African.

fw0stahl, 2015-07-14:
Ok thank you "set as default" is a workaround, haven't seen the button before :D

closed 2015-09-05 by hamish2014