hamish2014/FreeCAD_drawing_dimensioning #51: Missing numbers in drawing dimensioning preference setting.

iaminfinite, 2015-07-07:
Can't see variable numbers in Edit > Preference > Drawing dimensioning, as shown in attached image. But seem like plus and minus buttons works.
[image]
FreeCAD 0.15 rev 4671 on Fedora 22

hamish2014, 2015-07-08:
Hi iaminfinite, thanks for the posting the issue.
Question: are variable numbers only missing for *Edit>Preference>Drawing dimensioning* or are they also missing in other preference tabs as well, such as  *Edit>Preference>General*?

iaminfinite, 2015-07-09:
Hamish,

All others work fine, but Drawing dimensioning.

hamish2014, 2015-07-11:
Ai, I really do not know what the cause of your issue is.
On both my Windows and Ubuntu Linux FreeCAD's i do not experience the problem.
The Resources/ui/drawing_dimensioing_prefs-base.ui file is edited using Qt designer 4, do you think that is the cause the problem?
So i am afraid I can't be of much help.

In the mean while, adjust your dimensioning preference in the FreeCAD task dialog panel, as you create dimensions.
Changing the preferences there will have the same effect as changing them in Edit>Preferences

closed 2015-09-05 by hamish2014