hamish2014/FreeCAD_drawing_dimensioning #52: Running the Pyhton command failed

mgulbayrak, 2015-07-11:
Hi,

ı tried Freecad with dimensining tool but program always give same error.

[image]

hamish2014, 2015-07-11:
Hi  mgulbayrak, 

The Drawing Dimensioning workbench tools are for page objects generated using the Drawing workbench.
I have added an information window to tell users this.

closed 2015-07-11 by hamish2014

mgulbayrak, 2015-07-12:
Sorry , I missed it. 

Thank you.

averter, 2016-05-28:
Hi,

Either something is wrong with my Freecad build or I have some bug related with this error. 
Basically all of my drawing module commands only appear on the drawing dimensioning module (I don't know if this is normal?
Then whenever I try to add some dimension this error pops up. Even after having created a drawing.
[image]
Linux Mint 17.3 x64