hamish2014/FreeCAD_drawing_dimensioning #53: Dimension edit, no function for tolerance

fw0stahl, 2015-07-14:
Hello,

i have a problem to edit the whole tolerances on a worksheet. Maybe this failure could help you from the python-console:

could not convert string to float: <text x=Running the Python command 'dd_editText' failed:
Traceback (most recent call last):
  File "/home/mm2/.FreeCAD/Mod/FreeCAD_drawing_dimensioning/textEdit.py", line 95, in Activated
    maskBrush = maskBrush
  File "/home/mm2/.FreeCAD/Mod/FreeCAD_drawing_dimensioning/selectionOverlay.py", line 145, in generateSelectionGraphicsItems
    addSelectionPoint( *element.applyTransforms( float( element.parms['x'] ), float( element.parms['y'] ) ) )

hamish2014, 2015-07-14:
Hi  fw0stahl,

I have unfortunately been unable to replicate the issue.
Could you please either 
a) post the problem file on http://www.filedropper.com/ so that i can have a look at it, or
b) proved step-by-step instructions to replicate the issue.

fw0stahl, 2015-07-14:
The document is critical to send it to an other person, sorry. I look if i can reproduce the failure... in the moment new sketches are all right.

closed 2015-07-19 by hamish2014