hamish2014/FreeCAD_drawing_dimensioning #56: Possible bug in DrawingDimension WB?

Maavhamt, 2015-07-30:
Dear All,

I set the default dps to 0 in the DrawingDimension WB and things were going well until a dimension that should have been "70" came out "7". I set the dps to 1, redrew the dimension and it came out 70. Resetting to 0 dps it came out 7 again. Diagrams below. Possible bug or am I doing something wrong?

OS: Windows 8
Word size of OS: 64-bit
Word size of FreeCAD: 32-bit
Version: 0.15.4671 (Git)
Branch: releases/FreeCAD-0-15
Hash: 244b3aef360841646cbfe80a1b225c8b39c8380c
Python version: 2.7.8
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17

[image]

[image]

hamish2014, 2015-08-01:
Hi Maavhamt,

Thank you for your high quality bug report.
I have just patched the drawing dimensioning workbench, and the issue should be fixed now.
Please pull the latest version and check on your side

Best Regards,

Maavhamt, 2015-08-03:
Hi Hamish,

 

Thanks for the prompt response. When you say “pull the latest version” does that mean reload the whole DrawingDimension WB from GitHub via a ZIP file or is there a quicker way of updating it?

 

Thanks,

 

 

Regards,

 

 

Aidan Abrahart (alias “Maavhamt”)

___________________________________________________________
Aidan Abrahart

Applications Engineer
AG Instruments

Unit 57 | Avenue A | Stoneleigh Park | Kenilworth | CV8 2LG 

Mobile: +44 (0) 7989 649016

 <http://www.aginstruments.co.uk/> www.aginstruments.co.uk

 

AG Instruments

Technology made to measureTM                  

 

An ISO 9001:2008 Certified Company

hamish2014, 2015-08-03:
Hi Maavhamt,

If you followed the windows installation instructions given in Readme.md, then you will need redownload the entire workbench from Github via a Zip file and overwrite the previous file.

Maavhamt, 2015-08-03:
OK thanks I have reinstalled it and it works fine. thanks again for your prompt response. I have another question about the DrawingDimension WB which I have just posted on the FreeCAD Forum.

closed 2015-08-03 by Maavhamt