hamish2014/FreeCAD_drawing_dimensioning #58: Pyhton Scripting

Maaphoo, 2015-08-07:
This is a wonderful add-on. Thanks. Has anyone implemented the ability to access these functions using FreeCAD macros? If not, any recommendations on strategy for adding this functionality? Right now, we are primarily interested in center lines for holes, diameters and linear dimensions.

hamish2014, 2015-08-08:
Hi Maaphoo,

To my knowledge no-one has used the drawing dimensioning functions in FreeCAD macros.

To get you started, try the following in your python console, with a drawing page as the active tab and a bunch or ortho view with holes in them

```python

from dimensioning import getDrawingPageGUIVars
import selectionOverlay

V = getDrawingPageGUIVars()

CircleSelectionGraphicsItems = selectionOverlay.generateSelectionGraphicsItems( 
            [obj for obj in V.page.Group  if not obj.Name.startswith('dim') and not obj.Name.startswith('center')], 
            None,
            transform = V.transform,
            sceneToAddTo = V.graphicsScene,
	    doCircles=True, doFittedCircles=True,
            )

from centerLines import centerLinesSVG, d # where d is shorthand i use for the DimensioningProcessTracker instance

# retrieve the users preferences
preferences = dict( (p.name, p.getDefaultValue()) for p in d.preferences )

for cicleGraphicsItems in CircleSelectionGraphicsItems:
    scaling = cicleGraphicsItems.elementXML.scaling()
    x,y,r = [ cicleGraphicsItems.elementParms[k] for k in ['x','y','r'] ]
    svgCode = centerLinesSVG([x,y], [x+2*r,y+2*r], [x-2*r, y-2*r], scaling, **preferences)
    viewObj = App.ActiveDocument.addObject('Drawing::FeatureView','centerLines')
    viewObj.ViewResult = svgCode
    V.page.addObject( viewObj )

#now to clean up
from dimensioning import recomputeWithOutViewReset
recomputeWithOutViewReset(V)
```

Logic will need to added to determine which CircleSelectionGraphicsItems to create center lines for:
- often a circle is broken into multiple segments, each of which will have its own graphics item
- fillets, and other nuances  will need to be identified ...

I image that eventually you will have delve into the drawing dimensioning code, good luck and happy coding!

Maaphoo, 2015-08-08:
Excellent! Thanks hamish.