hamish2014/FreeCAD_drawing_dimensioning #59: French accent in page name make some problems

Cywil23, 2015-08-10:
Hi,

When I named a page with french accent "é", drawing dimensioning macro didn't work then.

Error message : 
Running the Python command 'dd_deleteDimension' failed:
Traceback (most recent call last):
  File "C:\Users\crpclb\AppData\Roaming\FreeCAD\Mod\Dimensioning\deleteDimension.py", line 56, in Activated
    V = getDrawingPageGUIVars()
  File "C:\Users\crpclb\AppData\Roaming\FreeCAD\Mod\Dimensioning\dimensioning.py", line 56, in getDrawingPageGUIVars
    pages = App.ActiveDocument.getObjectsByLabel( subWinMW.objectName() )

'ascii' codec can't encode character u'\xe9' in position 9: ordinal not in range(128)

hamish2014, 2015-08-12:
Ah, unicode:)
Please pull/download the latest version and let me know if your problem is resolved.

Cywil23, 2015-08-13:
Nice!
It's now ok with the lastest version :)
Thanks

closed 2015-08-13 by Cywil23