hamish2014/FreeCAD_drawing_dimensioning #61: Bug: Added text and grab point doesnt show up until new tool selected

arkticrainbow, 2015-08-28:
I am experiencing bug when adding text or grap point. Both act same: when I create drawing, add some views and make dimensions, everything works. Than I want to add text, click on desired location on drawing. Added text doesnt appear until I choose another tool (for example create linear dimension or edit text or any other) and close that tool too.

Video GIF attached will explain better.
[image]

This is Report View text regarding error it shows when adding text:

"
creating dimension dimText008
Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.15\Mod\DrawingDimensioning\previewDimension.py", line 133, in mousePressEvent
    removePreviewGraphicItems( recomputeActiveDocument=True, launchEndFunction=True )
  File "C:\Program Files\FreeCAD 0.15\Mod\DrawingDimensioning\previewDimension.py", line 90, in removePreviewGraphicItems
    recomputeWithOutViewReset( preview.drawingVars )
  File "C:\Program Files\FreeCAD 0.15\Mod\DrawingDimensioning\dimensioning.py", line 534, in recomputeWithOutViewReset
    selectionOverlay.hideSelectionGraphicsItems()
  File "C:\Program Files\FreeCAD 0.15\Mod\DrawingDimensioning\selectionOverlay.py", line 309, in hideSelectionGraphicsItems
    gi.hide()
RuntimeError: Internal C++ object (PointSelectionGraphicsItem) already deleted.
Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.15\Mod\DrawingDimensioning\dimensioning.py", line 446, in reject
    previewDimension.removePreviewGraphicItems( recomputeActiveDocument = True )
  File "C:\Program Files\FreeCAD 0.15\Mod\DrawingDimensioning\previewDimension.py", line 90, in removePreviewGraphicItems
    recomputeWithOutViewReset( preview.drawingVars )
  File "C:\Program Files\FreeCAD 0.15\Mod\DrawingDimensioning\dimensioning.py", line 534, in recomputeWithOutViewReset
    selectionOverlay.hideSelectionGraphicsItems()    
  File "C:\Program Files\FreeCAD 0.15\Mod\DrawingDimensioning\selectionOverlay.py", line 309, in hideSelectionGraphicsItems
    gi.hide()
<type 'exceptions.RuntimeError'>: Internal C++ object (PointSelectionGraphicsItem) already deleted.
"

hamish2014, 2015-08-28:
Thanks for the GIF.
Are you using the latest version?
I believed i patched this issue a couple of days back...

arkticrainbow, 2015-08-28:
I downloaded and rewrite old files in AddON - adding text and grab point now works as charm! Thank you very much.

(I had same issue with Assembly 2 AddOn when adding table to drawing - I had to close tool, choose another tool and close that too in order to show added table)

closed 2015-08-28 by arkticrainbow