hamish2014/FreeCAD_drawing_dimensioning #62: Multiple grab points on single line

arkticrainbow, 2015-08-30:
I got a question - maybe a bug. When I want to dimension create dimension on some object, there is multiple dots appearing for a single line, see image below:

[image]

I dont understand why so many grab points are created. I find it frustrating when I am having chamfer and need to grab the edges of it. I have to zoom in to find the grab point on the very edge of the line and than zoom out a whole lot to get to the other point.

[image]

Some other times mainly when I create not so complex shapes this doesnt happen. Is it as it suppose to be or is there something I can make about it?

hamish2014, 2015-09-05:
I appreciate your frustration, but the multiple dots originate from the drawing workbench.
In particular, the drawing dimensioning workbench parses the Svgs created by the drawing workbench.
In the case mentioned, the Svg created has multiple lines instead of just one straight line.
For the drawing workbench, I suspect that breaking the circular edge up into multiple lines helps with generating the drawing views ...

closed 2015-09-05 by hamish2014

arkticrainbow, 2015-09-05:
Fair enough than. Thanks for reply.