hamish2014/FreeCAD_drawing_dimensioning #63: Cross section + dimension lines/arrows ...

paul18fr, 2015-09-19:
Hi 

Ounce again I'm happy to use Hamish2014 development module. :+1: 

I do not have access to previous post, but I think it has ever been suggested:
- to be able to perform cross sections with hatching
- like the text/dimension values, to be able to move dimension lines/arrows (or after an upgrade / dimension change for example)

In any way, many thanks for your (great) work

Paul

hamish2014, 2015-09-23:
Thanks Paul, 
Cross-sections would be nice. 
As would be able to update dimensions after creation.

Unfortunately I have lots of stuff going on at the moment and am unlikely to get round at adding any of these enhancements.

efyx83, 2016-02-01:
Hi Hamish2014, you have made a great work with yours module to freecad.
It's been awhile since last question so I am requesting again for:
- detail,
- cross section,
- broken out section.
That would make freecad dimensioning mostly completed.
Is there any option you could work on this?
Best regards.