hamish2014/FreeCAD_drawing_dimensioning #64: Centers for arcs

Rafa350, 2015-09-28:
When create a linear dimension, the centers of arcs are not pressent. For circles the centers are OK

mghansen256, 2015-10-14:
The centers of arcs are needed for example when you have created a slot using the sketcher. Currently, you can only refer to the endpoints of the lines of the slot, but not to the centers of the arcs.

hamish2014, 2015-10-22:
I have added the requested feature.
I was hesitant to do so, since depending on the drawing being dimensioned, the user may not want arc centres to be visible.
i.e. arc centres for certain drawing may result in to much clutter.
But anyway, guess we will cross that bridge when we get to it.

mghansen256, 2015-10-24:
Thank you for implementing snapping to arc centers. Considering the clutter caused by too many points that you are worried about, would it be possible to re-use the snapping options used by the draft toolbar? Then the user could decide whether he would like to snap only to endpoints, arc centers, etc.