hamish2014/FreeCAD_drawing_dimensioning #66: Dimension points move and Dimensions disappear

OzSeats, 2015-10-17:
When attempting to create dimensions the points are highlighted in green (see attachment).
I can select the points and create the dimension but as soon as the dimension is created the points move off the drawing and the dimension disappears.

[image]

hamish2014, 2015-10-22:
Hi OzSeats,

It is rather Frustratingly when this happens.
On myside, the cause is that the drawing page's template property get mangled:
When the drawing pages template is a non-existent file, the drawing page wont update.

Hope you managed to figure the cause out  on your side, 5 days before this message.

mghansen256, 2015-10-24:
Here is a link to the FreeCAD bug report about the mangled template properties of drawing pages:

http://www.freecadweb.org/tracker/view.php?id=2184

hamish2014, 2015-10-25:
Thanks mghansen256.