hamish2014/FreeCAD_drawing_dimensioning #67: on windows V0.15 after restore the saved fcstd, FreeCAD will stop working if I switch workbench from start to any except drawing

qingfengxia, 2015-11-04:
The only way to make FreeCAD working with saved doc with drawing dim is:
after start-up, activate drawing workbench, then activate drawing dimensioning workench. 

Is that not a bug , but a procedure.  It relies on drawing workbench. so activate drawing workbench. 

Can dimensioning workench do a check and activate drawing workbench before activating it?

Thanks

hamish2014, 2015-11-05:
Pleasure qingfengxia.

Having to activate the drawing workbench every time before loading a file with drawing dimensions, sounds frustrating.

Luckily for me this is not required on my setup (Linux + FreeCAD 0.16)

Would it be possible for you check if the pre-activating of the drawing workbench is still required,
if you make use of FreeCAD 0.16?

qingfengxia, 2015-11-06:
Thanks, I tested with unstable 0.16.5352 dev x64  windows zip download from http://sourceforge.net/projects/free-cad/?source=typ_redirect, it is fine now.  It may be worth noting for windows user, as we usually will stick with 0.15.