hamish2014/FreeCAD_drawing_dimensioning #68: Failed to draw a dimension

arkticrainbow, 2015-11-12:
Dear,

I am having an issue with this addon. I have a 3D model, where I previously created drawing with 2D dimensions. Today I tried to add one new dimension to the drawing but failed.

Here is a summary from Report view:

Running the Python command 'dd_linearDimension' failed:
Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.15\Mod\DrawingDimensioning\linearDimension.py", line 255, in Activated
    **commonArgs
  File "C:\Program Files\FreeCAD 0.15\Mod\DrawingDimensioning\selectionOverlay.py", line 154, in generateSelectionGraphicsItems
    path = SvgPath( element )
  File "C:\Program Files\FreeCAD 0.15\Mod\DrawingDimensioning\svgLib_dd.py", line 203, in __init__
    self.arcs.append( SvgPathArc( element, _pen_x, _pen_y,  rX, rY, xRotation, largeArc, sweep, _end_x, _end_y ) )
  File "C:\Program Files\FreeCAD 0.15\Mod\DrawingDimensioning\svgLib_dd.py", line 313, in __init__
    assert dtheta >= 0

This happens with all tools in the addon. In some cases there are no anchor points shown at all, with some tools, there are only few, see picture below:

[image]

I tried to download latest DrawingDimensioning files - didnt helped. Any suggestion?

OS: Windows 7
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.16.5900 (Git)
Build type: Release
Branch: master
Hash: be5e79601a58e3f70194797acb8dd2f807a5f3ae
Python version: 2.7.8
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17

hamish2014, 2015-11-14:
Hi, 

Please upload your drawing page's Svg on http://www.filedropper.com/ (or the like) and i will have a look at it.
You can save your drawing pages Svg using Drawing workbench

arkticrainbow, 2015-12-09:
Dear hamish2014, sorry for long waiting, broken drawing was uploaded here: http://www.filedropper.com/brokendrawing