hamish2014/FreeCAD_drawing_dimensioning #75: dxf export not working

marcbritten, 2015-12-20:
Recently installed on windows 10 with Freecad 15.4671

Successfully created drawing and attempted to use the alternate dxf export you provide.

After selecting the filename and hitting OK, I get a red text on the lower status bar of freecad, but cann't read everything but it starts with

Running the Python command 'dd_exportToDxf' failed

hamish2014, 2015-12-23:
Hi,

Please post all the error text ( red text ) which you find in the report views ( View -> Views -> Report view)

marcbritten, 2015-12-26:
Running the Python command 'dd_exportToDxf' failed:
Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.15\Mod\DrawingDimensioning\unfold_export_to_dxf.py", line 20, in Activated
    export_via_dxfwrite( dxf_fn, V)
  File "C:\Program Files\FreeCAD 0.15\Mod\DrawingDimensioning\unfold_export_to_dxf.py", line 70, in export_via_dxfwrite
    from dxfwrite import DXFEngine as dxf

No module named dxfwrite

marcbritten, 2015-12-26:
I should add that the built in dxf export works (I've activated the library download thing)

marcbritten, 2016-01-10:
I've attempted to install dxfwrite a couple different ways and can't get Freecad to recognize it