hamish2014/FreeCAD_drawing_dimensioning #78: Not working when views are grouped on page ...

BPLRFE, 2016-01-10:
Hello.

Some sort of Bug recognized with 0.16.6189 on Windows 7 64-bit.
When having set up a drawing page and doing a right-click on "page" and selecting "create group",
after moving views to that group when selecting "add dimensions" no grab points will be displayed
on that view.
Would be great if that works or at least a "warning" would pop up ...

Renato-Rebelo, 2016-01-11:
I have the same problem, it is not possible to enter new dimensions.
I use the daily versions of Ubuntu and Windows
best regards,
Renato Rebelo

Renato-Rebelo, 2016-01-11:
I was testing and the problem is with the views made with arch bench or the draft bench if made directly with the Drawing from pieces no problem.
best regards,
Renato Rebelo

BPLRFE, 2016-01-11:
If you want to try with my files:
https://github.com/BPLRFE/Youtube-Tutorial-Models/tree/master/Tutorial%2020
The part was originally done with part-design, just made and used a simple copy for conveniance ...
The file with "1" at the end has grouped views - not possible to add dimensions to it (no grab points are shown).
The file with "2" at the end does not have grouped views, dimensions can be added ...