hamish2014/FreeCAD_drawing_dimensioning #81: "Edit dimension text" dissapeared

BPLRFE, 2016-01-10:
In older versions you could edit the dimesion's text to add diameter-symbol.
This seemed to have gone ...
Or was it just overlooked/ignored when updating the icons ?

There seems to be no way to edit the dimensions text via the data-tab ...

BPLRFE, 2016-01-11:
Same to Move dimension's text ... just dissappeared ...
I just rolled back to commit "c42b43d65f" and there both commands were present (with old icons)
and worked ...

liutas4x4, 2016-02-05:
The same:
On Mint (Unbuntu based) there are no Text Edit nor Text Move, while TextEdit.py and TextMove.py present in folder.

BPLRFE, 2016-02-18:
I did have a look at the code.
With the git commit mentioned by me above these commands were commented out BY PURPOSE.
Since in FreeCAD itself changing text-values and position would be done by selecting the object (here the dimension) and then selecting the data-tab and editing the text or applying a new placement. But it looks like this had not been programmed to full extent right now ...