hamish2014/FreeCAD_drawing_dimensioning #85: Detail, cross section, broken out section, hole table

efyx83, 2016-02-02:
Hi Hamish2014, you have made a great work with yours module to freecad.
It's been awhile since last question so I am requesting again for:

- detail,
- cross section,
- broken out section. That would make freecad dimensioning mostly completed. Is there any option you could work on this? Best regards.

What else should be very useful (and probably fast to do) is hole table. That means, you put a base on a view and automaticly is generated table with name, x, y, quantity and hole size position  (something like this [https://www.solidsolutions.co.uk/Uploaded/Image/Andy%20F/Hole%20Tag%20Order%20Modified%20Order.PNG](url)

hamish2014, 2016-02-02:
Hi  efyx83,

Unfortunately I do not have the time to add the requested features.

The drawing dimensioning workbench is open source, perpahs someone else can fork the code and add the requested features ...

ghislainvdwalle, 2016-03-06:
Hello efyx83,

good morning,

there is a workaround in overcoming the obstacle of cross sections and details.
You simply make the cross section in freecad & import it in the same drawing.
In the draft workbench you can give the cross section shades.
look at:
https://www.youtube.com/watch?=g2hWt_E2KMk&list=PLAYwpugSXElhhS1gxDA9WUWVLpy8-kF8e
In German language.

regards,