hamish2014/FreeCAD_drawing_dimensioning #88: Grab points not showing up

ebs675, 2016-03-06:
I can create a grab point, but they will not show up for use when trying to add a dimension.

ghislainvdwalle, 2016-03-07:
Hello ebs675,

good morning,

I can confirm the grab point is in the tab, but like you told, it is not showing up.

using 0.15 in Vista & 0.16 in Debian.

BPLRFE, 2016-03-08:
Well, rolling back to the version from 14th december did work. Grab Points showing up ...
So with the newer versions something must have been broken ...

ghislainvdwalle, 2016-03-09:
Hello BPLRFE,

good morning,

you didn't by accident shelve that 14 December version somwhere?
I'm trying to use this workbench on a daily basis in the shop right now.
Every improvement and workaround helps.

regards,

BPLRFE, 2016-03-09:
You can do this here on Github.
Make sure the code-view shows up with the drawing dimensioning workbench (that's the view you normally get here ...)
Click on "143 commits" and scroll down the list until you see the entry from 14th december
(Error handling added ...) and click on the right side on the commit number "3f5875b".
Now you see the changes the commit does. Click in the upper right corner on "browse files".
Now you get the code-view of the 14th-december-version.
Click on "Download ZIP" and replace the files on your Computer with the ones you
just downloaded ...

ghislainvdwalle, 2016-03-09:
thank you BPLRFE, changed it in a dash and works fine now.