hamish2014/FreeCAD_drawing_dimensioning #89: Drawing Dimensioning Workbench and Windows 10?

limitcycling-redstorm, 2016-03-08:
Hi Hamish,

I recently downloaded the drawing dimensioning workbench and set it up according to the instructions (I believe). It isn't appearing, however, on my workbench drop down. 

I am currently running windows 10 and my FreeCAD version data is below. 

OS: Windows 8
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.15.4671 (Git)
Branch: releases/FreeCAD-0-15
Hash: 244b3aef360841646cbfe80a1b225c8b39c8380c
Python version: 2.7.8
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17

Any assistance will be greatly appreciated. 

Limitcycling_Redstorm

pgilfernandez, 2016-03-15:
It neither works here on Windows 8.1, the report view tells me:
`During initialization the error DLL load failed: No se pudo iniciar la aplicaci�n; la configuraci�n en paralelo no es correcta. Consulte el registro de eventos de la aplicaci�n o use la herramienta sxstrace.exe de la l�nea de comandos para obtener m�s detalles. occurred in C:\Users\CEPATTL-1\AppData\Roaming\FreeCAD\Mod\drawing_dimensioning\InitGui.py`

Any idea?

aragocom, 2016-03-17:
Doesn't appear in the workbenches for me either - Windows 10 64 bit, FreeCAD 0.16 Revision 6597.
Report view shows "During initialization the error DLL load failed: The application has failed to start because its side-by-side configuration is incorrect. Please see the application event log or use the command-line sxstrace.exe tool for more detail. occurred in M:\Graphics\FreeCAD 0.16\Mod\DrawingDimensioning\InitGui.py".
I copied the folder in Mod under version 0.16 to the Mod folder under 0.15 and it appears in the workbench list and works OK when I run 0.15

pgilfernandez, 2016-03-17:
​Yes, it happens the same for me: it does work on v0.15 but not on v0.16...​

Pablo Gil Fernández
email:
Twitter:
pgilfernandez@gmail.com
@exkr

BPLRFE, 2016-03-21:
Did you guys try this
> make sure your previous attempt at installing is deleted
> use the FreeCAD-addons-installer here: https://github.com/FreeCAD/FreeCAD-addons
> install drawing dimensioning with the addon-macro

aragocom, 2016-03-21:
Deleted the DrawingDimensioning folders from the Mod folder in both 0.16 and 0.15. Installed and ran the macro which installed the folder drawing_dimensioning (note different spelling) in C:\Users\Peter\AppData\FreeCAD\MOD. No changes - 0.15 has it, 0.16 doesn't. As a check I deleted the  folder from the C:\Users\Peter\AppData\FreeCAD\MOD and 0.15 hasn't got it so it 0.15 is definitely accessing the workbench from the C:\Users\Peter\AppData\FreeCAD\MOD\drawing_dimensioning folder.

pgilfernandez, 2016-03-21:
It didn't solved the problem for me. I even tried deleting the parameter
folder created inside "Edit parameters" and, I don't know why, I think the
problem is related with this configuration...

Pablo Gil Fernández
email:
Twitter:
pgilfernandez@gmail.com
@exkr

BPLRFE, 2016-03-21:
Just confirmed.
Newest release doesn't show up on (virtual) windows10-install.
Workaround:
Roll back to version from 14th december. See my comment on [issue #88].

limitcycling-redstorm, 2016-03-23:
Thank you all for the feedback and ideas. I'll be trying some soon and will contribute my experiences here.

aragocom, 2016-03-23:
Rolled back to version 3f5875b from 14 December - no change, still not available. Installed files in both locations - in the user folder and the program folder. Is visible in 0.15.

BPLRFE, 2016-03-23:
That is odd. On my virtual Win10-System (it is the official VMWare-ISO of Microsoft) I am user "IEUser" and I downloaded the 0.16-dev-snapshot und unzipped it in the \downloads directory.
I installed the Runtime-Distributables and downloaded and unzipped drawing-dimensioning
workbench in the user folder. That did work.
Where is your 0.16-FreeCAD installed ? Maybe a problem with user rights ?
I am not quite sure what rights the user IEUser has, maybe admin priviliges ...

aragocom, 2016-03-23:
I have a drive dedicated to most of my programs - makes backup easy. So 0.16 is located in M:\Graphics\FreeCAD 0.16 - 0.15 under M:\Graphics\FreeCAD 0.15.. I suspected permissions when I first had problems so I created an Everyone group and gave it full control. Just checked - still full control for the folder and files while the folder C:\Users\Appdata\Roaming\FreeCAD\MOD\drawing_dimensioning has full control for me as a login. Note also that at the beginning I just simply copied the folder DrawingDimensioning from Mod under 0.16 to Mod under 0.15  and it worked perfectly in 0.15. I have no disabled workbenches in Tools > Customize > Workbenches. Note also that the error message in Report View (my first post) specifically mentions InitGui.py. I also installed another Workbench in 0.16 previous to Drawing Dimensions - just copied it to the Mod folder - and it showed up OK.

aragocom, 2016-03-25:
Firstly - I don't have a clue what I am doing ... However, poking around I decided to run the sxstrace program as suggested. First with the drawing_dimensioning folder present and then without. I noticed the ManifestPath with the folder present was FreeCAD 0.16\bin\lib\site-packages\numpy\core\multiarray.pyd while without it was FreeCAD 0.16\bin\FreeCAD.exe. Following a suggestion on a web page I used Vim editor to inspect the two files - in particular the part after <dependentAssembly>. In multiarray.pyd it is <assemblyIdentity type="win32" name="Microsoft.VC90.DebugCRT"  while in FreeCAD.exe it is <assemblyIdentity type="win32" name="Microsoft.VC90.CRT". So I backed up multiarray.pyd and edited out the Debug part. Still haven't got the Drawing Dimension workbench but now I don't get the "side-by-side configuration is incorrect" message. Instead I get "During initialization the error DLL load failed: %1 is not a valid Win32 application. occurred in AppData\Roaming\FreeCAD\Mod\drawing_dimensioning\InitGui.py". Thoughts ???

limitcycling-redstorm, 2016-03-26:
Good Morning all,

I followed the suggestion from BPLRFE, which I have quoted below and it worked exactly as described in the installation instructions for the addons macro. I now have what appears to be a working drawing dimensioning workbench.

Again I am running windows 10 and the following version of FreeCAD. 

OS: Windows 8
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.15.4671 (Git)
Branch: releases/FreeCAD-0-15
Hash: 244b3aef360841646cbfe80a1b225c8b39c8380c
Python version: 2.7.8
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17

Thanks for the assistance BPLRFE!
******
> 

Did you guys try this

make sure your previous attempt at installing is deleted
use the FreeCAD-addons-installer here: https://github.com/FreeCAD/FreeCAD-addons
install drawing dimensioning with the addon-macro

closed 2016-03-26 by limitcycling-redstorm