hamish2014/FreeCAD_drawing_dimensioning #90: Can not add DrawingDimensioning-Workbench within FreeCAD 0.16 (but 0.15)

mueller-christ, 2016-03-23:
Similar to bug #72 the drawing dimensioning workbench is not shown in the workbench selection of FreeCAD 0.16. Not successfully with FreeCAD versions 0.16.6560 and 0.16.6634. Neither downloading, extracting and copying the workbench to folder Mod, nor using the "FreeCAD Addons installer Macro" works. But it works properly with FreeCAD version 0.15.4671.

OS: Windows XP
Word size of OS: 32-bit
Word size of FreeCAD: 32-bit
Version: 0.16.6634 (Git)
Build type: Release
Branch: master
Hash: 297e84e3f064356d2a3403345e83a454a7eea923
Python version: 2.7.8
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17

OS: Windows XP
Word size of OS: 32-bit
Word size of FreeCAD: 32-bit
Version: 0.15.4671 (Git)
Branch: releases/FreeCAD-0-15
Hash: 244b3aef360841646cbfe80a1b225c8b39c8380c
Python version: 2.7.8
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17

Niteengk, 2016-06-18:
similarly, the Drawing workbench in 0.16 is not working . It can not create 2D views and error comes. on the views created.

rsxy, 2016-07-05:
For me, the drawing dimensioning workbench workbench does not appear in FreeCAD 0.16 stable (official download, Win 7/32), but it works with the 0.17 dev version.

OS: Windows 7
Word size of OS: 32-bit
Word size of FreeCAD: 32-bit
Version: 0.16.6704 (Git)
Build type: Release
Branch: releases/FreeCAD-0-16
Hash: 0c449d7e8f9b2b1fb93e3f8d1865e2f59d7ed253
Python version: 2.7.8
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17

microelly2, 2016-07-05:
hello rsxy
try the commands 

import assembly2lib
import dimensioning

on python console. it seem that numpy is not the right
see:
http://forum.freecadweb.org/viewtopic.php?t=8395&start=60#p77413

rsxy, 2016-07-05:
Hi microelly2, I tried that on the console included with FreeCAD, i.e.
"C:\Program Files\FreeCAD 0.16\bin\FreeCADCmd.exe"

and got this error, which also appeared for very short time at startup:

....
    import numpy.core.numeric as _nx
  File "C:\Program Files\FreeCAD 0.16\bin\lib\site-packages\numpy\core\__init__.
py", line 6, in <module>
    from . import multiarray
ImportError: DLL load failed: The application has failed to start because its si
de-by-side configuration is incorrect. Please see the application event log or u
se the command-line sxstrace.exe tool for more detail.

I copied the whole numpy tree (bin/Lib/site-packages/numpy) over, from 0.17dev to 0.16 - I don't know why, now it works in both versions. Thanks a lot!