hamish2014/FreeCAD_drawing_dimensioning #91: Vista and Win7 --drawing dimensioning not working.

ghislainvdwalle, 2016-04-25:


ghislainvdwalle, 2016-04-25:
Hi there,

good morning,

my system:
OS: Windows Vista
Word size of OS: 32-bit
Word size of FreeCAD: 32-bit
Version: 0.16.6700 (Git)
Build type: Release
Branch: releases/FreeCAD-0-16
Hash: 7b925d11aa69ac405b423635adb1e2833f18a817
Python version: 2.7.8
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17

I have started Freecad from the command line to look to the log (:FreeCAD -l)
and the log gives:

Log: Init:      Initializing C:\0\FreeCAD 0.16\Mod\FreeCAD_Drawing-Dimensioning(Init.py not found)... ignore (renaming the InitGui.py gives also a refusal)

Log: Init:      Initializing C:\0\FreeCAD 0.16\Mod\FreeCAD_Drawing-Dimensioning... failed

File "C:\0\FreeCAD 0.16\bin\lib\site-packages\numpy\core\__init__.py", line 6, in <module>
    from . import multiarray
ImportError: DLL load failed: The application has failed to start because its side-by-side configuration is incorrect. Please see the application event log for more detail.
Log: ----------------------------------------------------------------------------------------------------
Err: During initialization the error DLL load failed: The application has failed to start because its side-by-side configuration is incorrect. Please see the application event log for more detail. occurred in C:\0\FreeCAD 0.16\Mod\FreeCAD_Drawing-Dimensioning\InitGui.py

looking for help.

BPLRFE, 2016-04-25:
It seems like the libraries used by FreeCAD do not support Vista any more.
See here:
http://forum.freecadweb.org/viewtopic.php?f=8&t=15392&start=10
If you really need the vista-install i would try to install linux mint 17 via dual-boot and
use that os for FreeCAD

ghislainvdwalle, 2016-04-26:
thank you BPLRFE (once again) and good morning to you.
It is merely a nuisance, because I have the 0.16 on Debian 7.9 on another computer and over there the Drawing Dimensioning Workbench is A-OK.
I want to keep however a 386 system in the office because a lot of win pgms do not function properly on a WIN7 64 bit graphics environment.
I am a kind of an "oll guy" dragging with me even old but valuable MS-DOS pgms since DOS3.2.

thanks again and have a nice day.

closed 2016-04-26 by ghislainvdwalle