hamish2014/FreeCAD_drawing_dimensioning #97: Workbench failure( Freecad 0.16)

satheesh7, 2016-06-23:
registerPreferenceError: default for textFormat_radial redeclared

Khalinda, 2016-07-15:
I'm having the same problem when reopening a drawing that has a radial dimension.  This message pops up when attempting to load the Drawing Dimensioning workbench.  A side effect of this problem is that the Drawing Dimensioning toolbars fail to appear in the GUI, making it impossible to use the workbench on the problematic drawing.  

What I did, is delete the offending objects from the browser tree then exit FreeCAD.  Upon reopening FreeCAD, the drawing was left in a somewhat confused state, but I was able to load the Drawing Dimensioning workbench and restore the drawing to a proper state from there.

hamish2014, 2016-07-16:
Please note that I am disabling the Issue tracker for drawing_dimensioning.
For more than a year now I have not had the time to respond to any drawing_dimensioning issue properly (and wont be able to do in the future either).
I am therefore very uncomfortable leaving this issue tracker open, and am therefore closing it.
I hope that you have managed to resolve your issue in the meantime.

codeix, 2017-03-06:
I'am new in FreeCAD and was run into the same Problem. 
My Workaround:
1. Close and Reopen FreeCAD
2. Switch Workbench to Drawing Dimensioning
3. Open the .FCStd file

niel-malan, 2017-07-19:
@codeix 's workaround worked for me.