hamish2014/FreeCAD_drawing_dimensioning #98: Text editing, replacing dimension text and moving is totally dead on 0.16 version

liutas4x4, 2016-07-11:
Rolling back to FreeCAD 0.15 does not help: I see handles for text now, but text after editing or moving remain the same at the place it was created.

Khalinda, 2016-07-15:
I'm having the same problem (on versions 0.16 and 0.17...).  I'm not a developer, but the only thing that I can suggest is to revert to an earlier version of the workbench using GIT.  FreeCAD version 0.15 was released 2015/04/15, so likely the problem was caused by some change after that.  I see the following commits that modified textedit.py after that release date:

hamish 2015-06-03
hamish2014 2015-06-08
hamish2014 2015-06-11
hamish2014 2015-09-01
hamish2014 2015-09-04
hamish 2015-09-04

Khalinda, 2016-07-15:
It is at least possible to edit the text by selecting the object from the browser tree (it may take some trial and error to find the right dimension), then the text can be modified from the object properties.

liutas4x4, 2016-07-15:
While some kind of text editing is possible through data tab of object properties, dimensions itself do not have corresponding editable field. So there are no any workaround.
The second thing is that on git there are no older version of this workbench and it looks like author is bored with his idea.
Thanks the Lord, I found old download on my cloud. Rolled back. Will check soon combination of 0.16 and older version of DD.

hamish2014, 2016-07-16:
Please note that I am disabling the Issue tracker for drawing_dimensioning.
For more than a year now I have not had the time to respond to any drawing_dimensioning issue properly (and wont be able to do in the future either).
I am therefore very uncomfortable leaving this issue tracker open, and am therefore closing it.
I hope that you have managed to resolve your issue in the meantime.