hamish2014/FreeCAD_drawing_dimensioning #99: Re-licencing to LGPL-2.1

hamish2014, 2016-07-19:
Hi @yorikvanhavre @SebXcxl @crobarcro @ulikoehler 

When i created the FreeCAD_drawing_dimensioning I choose a licence which is incompatible with FreeCAD's licence.
Therefore I would like to change the licence to the GNU Lesser General Public License, version 2.1
This gives the FreeCAD team to option to use some of the code should they wish to.

As you are all contributors to drawing_dimensioning,  i need your permission to re-licence.
So please respond to this issue with your consent.

Thank you,

crobarcro, 2016-07-19:
I give permission for my contributions to be released under the GNU Lesser General Public License, version 2.1.

SebXcxl, 2016-07-19:
I give my permission too for my contributions to be released under the GNU Lesser General Public License, version 2.1.

yorikvanhavre, 2016-07-19:
I give my permission too for my contributions to be released under the GNU Lesser General Public License, version 2.1. (for assembly2 too if needed)

ulikoehler, 2016-07-20:
I hereby give permission for my contributions to be released under the GNU Lesser General Public License, version 2.1.

Sorry for taking so long to reply!

hamish2014, 2016-07-20:
Thanks everyone, much appreciated!
Will re-licence shortly

luzpaz, 2017-08-08:
@BPLRFE Can this issue now be closed?

closed 2017-08-10 by BPLRFE