hamish2014/FreeCAD_drawing_dimensioning #103: Pointer to describe features

steelman, 2017-07-02:
Please add a tool to draw a pointer to describe selected feature like [tapped holes](http://help.solidworks.com/2012/English/SolidWorks/sldworks/Cosmetic_Threads.htm).

BPLRFE, 2017-07-02:
Already possible.
Set a grab point at a location of your choice.
Use notation indicator tool to produce a note "1" withe circle.
Make sure the note is selected in the tree view.
Look at the data tab.
Set the "Note Circle_radius" to "0" (Zero).
Insert in field right to "note Text" the text you want to have displayed ...

closed 2017-07-02 by BPLRFE

steelman, 2017-07-02:
Thank you.

However, I've seen more obvious solutions to simple problems (-;

steelman, 2017-07-03:
Almost fine. Unfortunately the line ends right in the middle of the first character (centre of d=0 circle). Any leading whitespace characters in the text are stripped.