hamish2014/FreeCAD_drawing_dimensioning #106: Invalid comparison to None is centerLines.py

canismarko, 2017-09-11:
This line [1] causes and exception when placing centerlines on a drawing in with my version of FreeCad (0.16) and numpy (1.13.1) [2]. Replacing ``if bottomRight <> None:`` with ``if bottomRight is not None:`` fixed the problem for me.

[1] https://github.com/hamish2014/FreeCAD_drawing_dimensioning/blob/master/centerLines.py#L37

[2] https://gist.github.com/canismarko/15da5cf73b73d09d4dcf5520ddca2083

BPLRFE, 2017-09-17:
Thanks for mentioning. Please be more specific about your FreeCAD-Version-number.
In FreeCAD use the Help > About dialogue and copy the information to the clipboard (see button).
Paste it in here. Thanks ...

easyw, 2017-09-17:
In general would be correct to change all occurrences of '== None' and '!=None' and '<>None'  to 'is None', or is 'not None'

canismarko, 2017-09-18:
@BPLRFE, here's the version info you requested:

> OS: Linux
> Word size of OS: 64-bit
> Word size of FreeCAD: 64-bit
> Version: 0.16.Unknown
> Build type: Release
> Python version: 2.7.13
> Qt version: 4.8.7
> Coin version: 3.1.3
> OCC version: 6.9.1
>

quazgar, 2018-03-01:
Actually, I believe that Line 37 should rather be (if `BottomRight`is either None or a numpy array):
`if not bottomRight is None and bottomRight.all():`